andete/madparts #86: file management simplification

andete, 2015-07-20:
Get rid of all the file management, the files with uuid name etc. Just make it usable as a single footprint editor and use the filesystem for management.

spanner888, 2015-07-25:
+10 on this - I am havign problems creating new footprints. madparts seems to get confused and 
1. creates the new footprint code/file, but
2. ALSO overwrites one EXISTING madparts footprint code file with the new code template

Have had this happen a dozen times and can reproduce - have a bucket load of notes on this if you want detail

andete, 2015-07-26:
This is being implemented on the b2.0 branch.

closed 2015-08-12 by andete

spanner888, 2015-08-13:
madparts 2.0 - file handling
  - File menu currently has "save as", but no "file open" or "new". these would be really great additions to have!
  - Curently only way to open a file is to pass filename on cmdline. Suggest if no filename passed in, just open madparts without any code, or maybe default new template code.
  
Been running madparts on my new Novena laptop with several libraries open and working on a complex footprint, so it is very obvious that the new file management code is much faster (without overhead of compiling all the added libraries?)

andete, 2015-08-13:
I didn't add a 'new' as my usual way of working is open a design, do a save-as and then make the new part. En empty footprint as a template seems not very useful.
An 'open' actually makes some sense, although e.g. on Windows you just double-click on a .coffee and it just works. On Linux  just close the window and open a new footprint from the shell.
I'll make a separate task for the 'open' as I think that is a good idea, and quite easy to add.

andete, 2015-08-13:
I've changed the code to have it offer the 'open' dialog when started without argument.

commit 8da201e5 (2015-08-12): implement file management simplification (Close #86)