andete/madparts #99: Exporting in Windows

ahatstat, 2015-12-06:
I am not able to export footprints in Windows 10 to either kicad old or new version.  The GUI opens and parts compile and are displayed fine.  The same error occurs with the GUI and the command line.  Here is the error when using command line.  I tried a few of the example files with the same result.  

C:\MadParts\madparts-2.0.1>C:\MadParts\madparts-2.0.1\madparts-cli.exe export C:\Users\Andrew\Documents\KiCad\AL_Footprints\coffee\DO-214AA.coffee C:\Users\Andrew\Documents\KiCad\AL_Footprints\AL_Diodes_SMD.pretty

Qt: Untested Windows version 6.2 detected!
DO-214AA compiled.
Traceback (most recent call last):
  File "madparts-cli", line 22, in <module>
  File "main\cli.pyc", line 130, in cli_main
  File "main\cli.pyc", line 32, in export_footprint
  File "export\kicad.pyc", line 70, in export_footprint
  File "mutil\mutil.pyc", line 21, in eget
Exception: Id not found

ahatstat, 2015-12-06:
I fixed the problem by adding `#id 1` to the header section  of my coffee file
Example file SOT23-5.coffee also doesn't have an id listed and fails the same way.  
Is this a bug or is the ID tag required?

andete, 2015-12-07:
Same issue as #94 .
This has been fixed but a new release has not yet been made.