andete/madparts #101: madparts crashes on Windows 8.1

Khalinda, 2016-01-11:
I downloaded madparts yesterday, and it has already crashed on me at least a half dozen times.  It crashes due to an access violation, apparently at the same location each time.  I would attempt to track down the cause, but I know next to nothing about Python.

I tried to do a so-called build of madparts based on the instructions provided at github, but when I attempt to run madparts.py, it complains that gui.dialogs is missing.  What library/extension would that be in?

andete, 2016-01-11:
Hello,

The build instructions for win32 are not up to date i seems.
You just need to run the "madparts" script in the root directory with python.
I mean:

C:\python27\python.exe madparts

instead of:

C:\python27\python.exe madparts.py

That being said, I've been using Linux for many years now so my knowledge of Windows is limited.

Joost

Khalinda, 2016-01-11:
Thanks Joost, that works.

closed 2016-09-27 by Khalinda